aligator/GoSlice #9: simple temperature gcode

aligator, 2020-05-20:
add simple gcodes for the temperature:
* temperature for the first layers (configuration for how many)
* temperature for all layers above the first layers

For both, hotend and heated bed.

https://marlinfw.org/docs/gcode/M104.html
https://marlinfw.org/docs/gcode/M109.html
https://marlinfw.org/docs/gcode/M140.html
https://marlinfw.org/docs/gcode/M190.html

Also turn them off at the end.

masihtehrani, 2020-05-21:
@aligator 
I'd like to work on it
can you more explain about the issue

aligator, 2020-05-21:
Great,
currently I have no temperature commands at all.
The plan is to have a M109 and M190 command in the starting GCode which sets (and waits for) a target  temperature (which should be configurable).

Then after it reaches a (configurable) layer, it should use the M104 and M140 to set the temperatures for the higher layers.
All 4 temperatures should of course also be configurable.

You can also suggest own ideas for temperature handling.
Do you have a 3D printer to test if the commands work as expected (haven't tried them yet, just looked them up in the marlin docu)?

To get startet, I suggest you to add the configuration values here:
https://github.com/aligator/GoSlice/blob/master/data/option.go
* to the struct (in this case I think the FilamentOptions struct)
* to the default config
* to the ParseFlags method

Then you have to inject the GCode.
For now I think it's enough to add it to the https://github.com/aligator/GoSlice/blob/master/gcode/renderer/layer.go PreLayer renderer similar to the fan I start there (M106).

The configured values can be obtained there using the options parameter. 

If you have any idea for improvements, please create issues  :-)

masihtehrani, 2020-05-21:
> Great,
> currently I have no temperature commands at all.
> The plan is to have a M109 and M190 command in the starting GCode which sets (and waits for) a target temperature (which should be [...]

Oh sorry I don't have a 3D printer and I don't have knowledge about this device. I just know Golang. I don't know I can help you on this issue or not

aligator, 2020-05-21:
ok, that's no problem.
Just try it, I will test it then. 

But you may need to learn a bit how GCodes work, it's not that hard, it's basically a simple list of commands.

If you need a viewer for the gcode to check it, I can suggest the Cura slicer. It can also open GCodes and you can configure it to show which temperature is used by colorizing the model.
https://ultimaker.com/de/software/ultimaker-cura

If you have questions, just ask.

aligator, 2020-06-06:
done by https://github.com/aligator/GoSlice/commit/c3377c69cd0175f3e4cc57d8f08cc55e6cd5b6c2

closed 2020-06-06 by aligator