aligator/GoSlice #17: configurable start and end gcode

aligator, 2020-06-06:
Currently the start and end gcode is hard coded [here](https://github.com/aligator/GoSlice/blob/master/gcode/renderer/layer.go).

It would be great to have a configurable start and end gcode.
For now it would be enough to just use two string-options which are just injected into the gcode.

kpfaulkner, 2020-06-30:
Is this in regard to the non instruction messages?
ie.

"Generated with GoSlice"
"______________________"

Just want to be able to customise it?

aligator, 2020-06-30:
I think this should be left hardcoded as a sort of branding.
> b.AddComment("Generated with GoSlice")
> b.AddComment("______________________")

With this issue I meant this block:
```go
		// starting gcode
		b.AddComment("START_GCODE")
		b.AddCommand("G1 X0 Y20 Z0.2 F3000 ; get ready to prime")
		b.AddCommand("G92 E0 ; reset extrusion distance")
		b.AddCommand("G1 X200 E20 F600 ; prime nozzle")
		b.AddCommand("G1 Z5 F5000 ; lift nozzle")
		b.AddCommand("G92 E0 ; reset extrusion distance")
```
and for the ending gcode this:
```go
		b.AddComment("END_GCODE")
		b.SetExtrusion(options.Print.LayerThickness, options.Printer.ExtrusionWidth, options.Filament.FilamentDiameter)
		b.AddCommand("M107 ; disable fan")

		// disable heaters
		b.AddCommand("M104 S0 ; Set Hot-end to 0C (off)")
		b.AddCommand("M140 S0 ; Set bed to 0C (off)")
```

kpfaulkner, 2020-06-30:
Should something like setting temperatures be something purely user defined? For safety shouldn't that always be executed? (yes, there is the argument about user knows what they're doing... but it would be easy enough to miss).

aligator, 2020-06-30:
Good point, but I am more a fan of customizability and good defaults.

Another way would be something like an additional flag "ForceSafeDisable" or something like that which is set to true as default and adds the safe gcode after the custom gcode.
It would be a bit more complicated but maybe a good compromis...

What do you think?

aligator, 2021-10-19:
Hi  :-)
I also think that PrinterOptions is better, because in most cases it is printer specific.

Just a note for you:
I am planning to change the license. So if you contribute, make sure you are also ok with the new license (Apache 2). I only need one missing agreement to finally change it: https://github.com/aligator/GoSlice/issues/44