aligator/GoSlice #18: simple fan control

aligator, 2020-06-06:
Similar to the temperature control we need a fan control, which starts the fan at a configurable layer with a configurable speed. This can be done [here](https://github.com/aligator/GoSlice/blob/master/gcode/renderer/layer.go). 

Currently the fan is hard coded to run at 100% starting at the third layer.

kpfaulkner, 2020-06-27:
Hi @aligator 

Looking at submitting a PR for this, but (after discussing with my 3D printing son, @williamf1), the question about do we want a single fan level or multiple?

Say, should we have something like layer X==fan speed XX%, layer Y == fanspeed YY%  etc?
From a command line argument point of view, there are a few options but nothing super neat.

option 1:  --fan-speed <layer|fan value, layer|fan value> 
option 2:  --fan-start-speed XXXX  --fan-increment-per-layer Y

(just 2 random examples).

Thoughts?

Ken

aligator, 2020-06-27:
Hi @kpfaulkner 

the initial issue was meant for a simple control, but I am fine with a more complex one also, great idea :-)
Later we may extend this for other features also, like e.g. configurable temperature for different layers (useful for heat towers).
I think Simplify3D can do something like this.

The command line arguments are currently kept simple, because i set the focus on the slicing process. (and later we may have a GUI?) So I think for now something like this may be enough: (similar to your option 1)
--fan-speed 3=50,10=100

You can decide on your own the exact look, as you think it's best. But I would avoid spaces, because then you just can get one string and parse it. With spaces either escaping would be needed on the command line or you get them as separate arguments, which would be harder to handle.

I am looking forward to your PR :-)

aligator, 2020-06-30:
Thank you for your contribution :+1:

closed 2020-06-30 by aligator