aligator/GoSlice #22: [Suggestion] Sequence of articles/tutorials?

dector, 2020-06-29:
Have you thought about documenting decision made in project or writing follow-up tutorials sequence. Simple tiny slicer can be a great medium-hard educational project but it's quite hard to find lots of information about used algorithms or tips-n-tricks.

Thanks!

aligator, 2020-06-29:
Yes, I've already thought about creating at least an explanation of the basic functionality, among other things because there is actually very little information about it and because it might also lower the entry hurdle for someone who wants to help with the project.

The only thing I found was actually this: https://www.instructables.com/id/How-to-Build-Your-Own-3D-Printing-Slicer-From-Scra/ which I didn't even read through completely because this Autodesk Dynamo Studio is too abstract and too far away from real code.
And then I found the first commit from Cura, which even daid303 himself pointed out in the comments here: https://hackaday.com/2017/03/01/diy-3d-slicer-is-a-dynamo/
This was the best I found, although large parts of the code are not too much commented.

Last but not least you can find some infos about the functionality of slicers here: https://tams.informatik.uni-hamburg.de/publications/2018/MSc_Daniel_Ahlers.pdf

I can't really say anything about the used polygon algorithms, because I'm using the Clipper Lib, which is more of a black box for me.

But of course, documenting is also connected with time, of which I, like most people, don't have that much. Therefore I would be glad about some help here, too. 

I can start to write my first thoughts and steps together. 

In any case thank you very much for the suggestion.

Translated with www.DeepL.com/Translator (free version)

dector, 2020-06-29:
Thanks a lot for provided links! I also found [this B.Sc. work](http://blog.hk-fs.de/wp-content/uploads/2015/08/Paper_Fabian_Schurig_3D_Printing.pdf) - it has some Java code but not that much information. But it's quite difficult to read about edge-cases and pitfalls without diving into sources of existing slicers.

aligator, 2020-06-29:
I started some overview-documentation here: https://github.com/aligator/GoSlice/tree/master/docs
Also there are many comments in the code also for the parts which are from the Cura port which I commented while porting it to understand it better myself. 

In the end I think most documentation will still be directly in the code, as the `go doc` tool is very great and I see no point in documenting it again in a different place. But for such an overview and lessons learned docu it's ok :-)

dector, 2020-06-29:
That's great, thank you!