aligator/GoSlice #26: Slicing failed with error:  error while calculating the union of new infill with already existing one

tobychui, 2020-08-20:
Testing with the following file downloaded from the internet (ASCII stl)
[test.zip](https://github.com/aligator/GoSlice/files/5101637/test.zip)

which shows the following error when I am trying to slice it
[image]

and I also noticed that the stl get overwritten by GoSlice when slicing failed, not sure if it is a bug or feature.

I tried to check the version of GoSlice but it seems neither do --version nor --v gives any versioning info. Maybe you can add in a command to check the version info as well?

Lastly, I love your work and find it amazing, add oil :)

aligator, 2020-08-20:
Thank you, I will take a look :-)

Also the version command is a good idea.

aligator, 2020-08-20:
> and I also noticed that the stl get overwritten by GoSlice when slicing failed, not sure if it is a bug or feature.

That is because till release 0.0.2 it always generated a test.stl file with the optimized stl for debugging.
So if you do not name it 0.0.2 it won't get overwritten.

This will be removed in future versions, I commented this out in the branch I am currently working on.

aligator, 2020-08-20:
should be fixed with https://github.com/aligator/GoSlice/commit/8bd9432f1aeb3834d2ff6de4960682e38e5db72d
also I disabled the test.stl output in the master. Could you please test it again?