aligator/GoSlice #42: Web UI based on GoSlice

tobychui, 2021-04-18:
Hi! I have created a Web GUI for GoSlice to be used in my cloud system called "SlicerA". That is why I am sharing it with you using this "issue" (as you don't seems to leave any email or threads specially for sharing downstream projects). You can check it out over here: https://github.com/tobychui/SlicerA

[image]

The SlicerA is also open sourced and licensed under AGPL. As this is my first time integrating an open source software based on AGPL, please let me know if there are any problems in my way of crediting the author under the AGPL requirements.
 
Have a nice day :)

aligator, 2021-04-18:
Hi, very cool! 

**AGPL:**
Basically as long as you publish the source as AGPL everything is fine. The only thing which is different from GPL is that you cannot Host any altered GoSlice version without providing the sourcecode. ->  `Network use is distribution `

And that's basically what I want to achive with AGPL: If you host it somewhere (cormmercial or not) I want the sourcecode back if it was altered, and I want to  know that GoSlice is "in" it. ->  License and copyright notice

So for your project this basically means: For the binaries you provide in the repo, I think your notice https://github.com/tobychui/SlicerA/blob/main/goslice/README.md here is sufficient (if you didn't alter the source. In that case just provide the source for the changes).  
Also it is a good idea that you ship it as a separate module because of the copyleft. 

-> so everything fine :+1: 

**About your project:**
I also wanted to create a web gui, but couldn't get to it yet because of many other (mostly unfinished) projects.  
Its very cool that you started one :-)
The only thing I did was experimenting with running GoSlice as webassembly https://aligator.dev which works, somehow, but is not very fast and blocks the browser js thread most of the time. (Just type `goslice gopher.stl` or any direct link to another stl file e.g. from thingiverse)

As I modified GoSlice since our last discussion to be better usable as Go lib, maybe it would be an idea to leverage that instead of shipping a GoSlice binary. https://github.com/aligator/GoSlice#use-as-lib
I also used that for the webassembly version https://github.com/aligator/dev/tree/main/go/goslice

In the end it would be nice if SlicerA would be also runnable without ArozOS (very cool project by the way :-) as standalone web-app. But that may be an extra project which just shares some common code with SlicerA. What do you think?

Thanks for sharing :+1:

tobychui, 2021-04-21:
Nice to know my crediting in the SlicerA is complying the AGPL 👍

> In the end it would be nice if SlicerA would be also runnable without ArozOS

Sure, making SlicerA run-able in standalone mode will be good for those who don't know how to use ArozOS and just want a simple web slicer. I guess I will need to schedule some free time next week and try to trim out something in standalone mode as AGPL license is a bit complex to deal with.

aligator, 2021-04-21:
I also started experimenting with threejs https://github.com/aligator/dev/blob/gcode-viewer/src/gcode/gcode.ts Maybe this will be a working gcode viewer lib some day :-) 

Some thoughts to AGPL: I am also not that experienced with it. Initially I chose it because I rewrote the initial commit of Cura as starting point and therefore "used" Cura code which is AGPL. Also I think it fits to what I want: open source everything what anyone does with GoSlice, even if it is just hosting a (modified) version.

As far as I understand it, you could create a GoSlice server using GoSlice as lib in an extra Project as AGPL which exposes for example a REST api. 
Then you could call this server from any frontend and the license of the frontend doesn't matter, as it doesn't include any GoSlice source.

If you host such a server you would need to provide the source of the GoSlice Server (as AGPL) and give credits that the frontend uses it. But the Frontend can be whatever License you want, as it just calls the api.

As I am not a lawyer, this may be wrong ;-) It's really not the easiest topic...

tobychui, 2021-04-24:
I do not think rewriting Cura's code by hand fitted the "used" definition in license term. In my opinion, as soon as it is not machine translated or parts that are copied directly from the upstream (Cura) source, you can claim it as yours even the logic are similar or identical. 

I have find an alternative STL viewer that is MIT Licensed and integrated with the new SlicerA "standalone mode". Feel free to have a look at it when you have the time to do so :)

aligator, 2021-04-24:
yeah maybe
 Anyway I still like the prinziple of GPL + Network use is distribution :-)

Very cool, thanks, I will check it out 👍

tobychui, 2021-04-24:
Maybe I will be closing this issue now. If you find any bugs or feature wanted on SlicerA, feel free to open an issue on the SlicerA repo. Thanks!

closed 2021-04-24 by tobychui